kipr/botui #33: Building botui on non-Wallaby hardware

PhilipTrauner, 2017-04-16:
I'm currently trying to compile botui on an Ubuntu 17.04 machine (I couldn't determine if Debian or Ubuntu is the main dev-target). 

These are the dependencies I could figure out:
`apt-get install qt5-default qt4-dev-tools build-essential cmake libssl-dev libboost-dev libpng-dev zlib1g-dev libbson-dev libbson-1.0-0 libopencv-dev doxygen python-dev swig libjpeg-dev libzbar-dev`

After installing (in that order)
1. libkar
2. pcompiler
3. bsonbind
4. daylite
5. libaurora
6. libwallaby
7. libbotball

and copying over `turn_off_wallaby.wav` from a Wallaby update I could successfully build botui.
Problem is that it segfaults immediately. On a different system (Debian 8 / armel / a Wallaby) it segfaulted after printing out `realloc(): invalid pointer` followed by a random address.

A strace on the Ubuntu system revealed following details:

**libwallaby.so** was located:
```open("/usr/lib/libwallaby.so", O_RDONLY|O_CLOEXEC) = 3```

interestingly it is searched for in weird places:
```
open("/lib/x86_64-linux-gnu/tls/x86_64/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/x86_64-linux-gnu/tls/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/x86_64-linux-gnu/x86_64/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/x86_64-linux-gnu/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/x86_64-linux-gnu/tls/x86_64/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/x86_64-linux-gnu/tls/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/x86_64-linux-gnu/x86_64/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/x86_64-linux-gnu/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/tls/x86_64/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/tls/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/x86_64/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/tls/x86_64/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/tls/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/x86_64/libwallaby.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libwallaby.so", O_RDONLY|O_CLOEXEC) = 3
```

**USB ports** are read for some reason:
```
readlinkat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:06.0/usb2", 0x559cc750ae00, 99) = -1 EINVAL (Invalid argument)
readlinkat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:06.0/usb2/2-1", 0x559cc750ae00, 99) = -1 EINVAL (Invalid argument)
readlinkat(AT_FDCWD, "/sys/devices/pci0000:00/0000:00:0b.0/usb1", 0x559cc750aa70, 99) = -1 EINVAL (Invalid argument)
```

I really don't know how to continue. I have already looked at the bitbake recipes which pretty much do the same things I have described. 

I would really appreciate help :)

bmcdorman, 2017-04-16:
Does a `gdb` backtrace or `valgrind` memcheck reveal the location of the segfault?

PhilipTrauner, 2017-04-16:
**gdb** output:
```
(gdb) run
Starting program: /usr/local/bin/botui
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffd95fd700 (LWP 1865)]

Thread 1 "botui" received signal SIGSEGV, Segmentation fault.
0x00007ffff3c85ec7 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
```

**valgrind** output:
```
==1866== Invalid read of size 2
==1866==    at 0x8B1DEC7: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.7.1)
==1866==    by 0x8B24604: operator<(QString const&, QString const&) (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.7.1)
==1866==    by 0x4E51C3E: bool qMapLessThanKey<QString>(QString const&, QString const&) (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E5297F: QMapNode<QString, Compiler::Base const*>::lowerBound(QString const&) (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E520DB: QMapData<QString, Compiler::Base const*>::findNode(QString const&) const (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E5161C: QMap<QString, Compiler::Base const*>::find(QString const&) (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E50F83: Compiler::Compilers::addCompiler(Compiler::Base const*) (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E5F07E: __Board_register() (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4010819: call_init.part.0 (dl-init.c:72)
==1866==    by 0x401092A: call_init (dl-init.c:30)
==1866==    by 0x401092A: _dl_init (dl-init.c:120)
==1866==    by 0x4000CD9: ??? (in /lib/x86_64-linux-gnu/ld-2.24.so)
==1866==  Address 0x46e0f77a is not stack'd, malloc'd or (recently) free'd
==1866==
==1866==
==1866== Process terminating with default action of signal 11 (SIGSEGV)
==1866==  Access not within mapped region at address 0x46E0F77A
==1866==    at 0x8B1DEC7: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.7.1)
==1866==    by 0x8B24604: operator<(QString const&, QString const&) (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.7.1)
==1866==    by 0x4E51C3E: bool qMapLessThanKey<QString>(QString const&, QString const&) (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E5297F: QMapNode<QString, Compiler::Base const*>::lowerBound(QString const&) (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E520DB: QMapData<QString, Compiler::Base const*>::findNode(QString const&) const (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E5161C: QMap<QString, Compiler::Base const*>::find(QString const&) (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E50F83: Compiler::Compilers::addCompiler(Compiler::Base const*) (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4E5F07E: __Board_register() (in /usr/local/lib/libpcompiler.so)
==1866==    by 0x4010819: call_init.part.0 (dl-init.c:72)
==1866==    by 0x401092A: call_init (dl-init.c:30)
==1866==    by 0x401092A: _dl_init (dl-init.c:120)
==1866==    by 0x4000CD9: ??? (in /lib/x86_64-linux-gnu/ld-2.24.so)
```
The output of gdb would lead me to believe that botui does not play well with qt5. Problem is, that both libkar and pcompiler depend on qt5 and linking with them only works when botui is built with qt5 **and** qt4 installed.

jsoutherland, 2017-04-16:
We don't have qt5 available in bitbake, so we have been using the qt4 branches of our dependencies (https://github.com/kipr/pcompiler/tree/use_Qt4)

@bmcdorman will be removing pcompiler as a dependency of botui soon but it may not be soon enough for your project.

jsoutherland, 2017-04-16:
oh, and libkar: https://github.com/kipr/libkar/tree/use_Qt4

PhilipTrauner, 2017-04-16:
Thanks for the quick responses, will try building with qt4 exclusively tomorrow.

PhilipTrauner, 2017-04-17:
**libkar** is seemingly still used by pcompiler, even in the "use_Qt4" branch
[This](https://github.com/kipr/pcompiler/blob/master/tools/pco.cpp) is the file in which kar is still included. I guess it's the command line component of pcompiler.

```bash
Linking CXX shared library ../lib/libpcompiler.so
[ 97%] Built target pcompiler
Scanning dependencies of target pco
[100%] Building CXX object tools/CMakeFiles/pco.dir/pco.cpp.o
/root/pcompiler/tools/pco.cpp:8:23: fatal error: kar/kar.hpp: No such file or directory
 #include <kar/kar.hpp>
```

```bash
> git branch
* use_Qt4
```

It successfully compiles with libkar installed, though.

With that being said, botui ran successfully for the first time. One problem remains, though: All buttons and dropdown are black, I suspect a missing font.

navzam, 2017-04-17:
If we aren't using libkar for user programs anymore, then I doubt the pcompiler command line tool is useful. Tools compilation should probably be an option anyway.

PhilipTrauner, 2017-04-17:
Before closing this issue I still have some remaining questions and observations:
* Buttons become properly coloured only if botui is launched as a subprocess (I have no idea how that's even possible)
* The fullscreen toggle doesn't really do anything and the standard window size is the one hardcoded in the ui templates (320x240). After going through the Wallaby updates I could not locate how the window size is externally set, did I miss a X11 setting somewhere?

Thanks for all the help so far 😄

jsoutherland, 2017-04-17:
@PhilipTrauner  I think botui is usually small.  We don't really use it outside of the controllers.

You can make the window larger here:
https://github.com/kipr/botui/blob/master/src/RootController.cpp#L103-L104
(but it doesn't scale components... it will help with larger camera previews)

On a computer you will want to force the cursor visible here:
https://github.com/kipr/botui/blob/master/src/CursorManager.cpp#L11
(we need a setting in botui for that...)

PhilipTrauner, 2017-04-17:
@jsoutherland The resolution of the Wallaby screen ([this one](https://store.gumstix.com/newhaven-43-capacitive-multi-touch-lcd-screen.html) I guess) is 480x272, that's why I'm confused. Isn't 320x240 the resolution of the Link screen?

(I'm running botui on actual Wallaby hardware, just not under a bitbake-built OS, but Debian 8 instead.)

jsoutherland, 2017-04-17:
That is a good question... I'll have to look into it.

jsoutherland, 2017-04-18:
@PhilipTrauner 
I don't have the answer yet...
The link resolution is 320x240 and the wallaby screen is 480x272 (resistive touch variant equivalent of what you linked).  I verified that the display is running at 480x272.
The display may be fullscreened and I am missing where that is happening.  It is probably in X as the terminal is also scaled to use the entire screen.

PhilipTrauner, 2017-04-18:
@jsoutherland I'm guessing that xserver-nodm (the one shipped with bitbake) actually comes with a very slim display manager, that would also explain why the window title is shown for the terminal. I'll change the resolution manually tomorrow and report my findings.

PhilipTrauner, 2017-04-19:
**Update**: We've forked botui and upped the default display resolution, should I open a pull request?

jsoutherland, 2017-04-19:
Yeah that sounds good. Thanks.

PhilipTrauner, 2017-04-20:
@jsoutherland Everything seems to be working great now, going to close this issue. Thanks for the help, really appreciate it! 😄

closed 2017-04-20 by PhilipTrauner