kipr/botui #54: A,B,C Buttons

aborgerding, 2019-09-10:
From Charles - The ABC buttons for a running program are too low on the screen to fully display

This doesn't seem to be an issue on our end. Attaching images showing what we see from the controller. Waiting additional feedback from Charles. 

These images are from 24.8:
[image]
[image]

aborgerding, 2019-09-10:
Charles - Here they are.  It appears the title bar is pushing them down.  I took these after a reboot to make sure the effect isn't a leftover.  When I get a chance I'll do a fresh image to see if it persists, which if so would indicate something is amiss with my particular Wombat.

[image]
[image]

Potentially result of previous testing. Waiting to see update from Charles after refresh.

Zacharyprime, 2020-02-28:
This issue has not been present in production yet and appears to have only been a problem in the development stage (or if the user is messing around with Debian desktop controls). I think this was just an issue with Charles' Debian settings or his image but the current standard image does not have this issue.

If somebody is able to recreate this issue and show any of the active developers, we can reopen it.

closed 2020-02-28 by Zacharyprime