kipr/botui #58: Turn OFF WiFi button doesn't operate as intended

prithvirajkadiyala, 2020-01-29:
Pressing Turn OFF Wifi, turns off WiFi but doesn't update to Turn ON WiFi.

Zacharyprime, 2020-03-12:
It's actually been a feature for tech support. If we do fix the button, it should be hidden behind a "Advanced" tab unless it functions more efficiently than the Wallaby's OFF button. 

The OFF button on the Wallaby was slow to react and update the UI. Paired with the fact that the WiFi turned on ~5 minutes after the Wallaby booted (or someone hit the ON button), caused students and teachers to try to turn the WiFi off and on to fix an issue that didn't exist and usually ended up messing something up. For some reason, turning the WiFi "off" prematurely on the Wallaby would cause the WiFi to act sporadically or stop working completely without rebooting the controller.

prithvirajkadiyala, 2020-03-27:
Got the Button working, We can turn the WiFi On or OFF from the Network page.

closed 2020-03-27 by prithvirajkadiyala