kipr/botui #63: Select AP or Client Mode

justinmerrell, 2021-02-09:
Upon boot, the controllers are set up and configured to be an access point. Ideally the option to connect to a network as a client, therefore having internet access would open the door to OTA updates and integrations such as Github. 

This thread will be used as documentation to work backward identifying how the network is currently configured along with other relevant information. If there is a  better location for this please let me know.

justinmerrell, 2021-02-09:
Old paper I dug up that appears to still be relevant: https://manualzz.com/doc/24576535/hacking-the-wallaby---practical-robotics-institute-austria

Zacharyprime, 2021-02-24:
I would like to add a more complex WiFi wizard that would allow for switching.
The default of AP mode needs to remain the same.

It needs to be hidden behind something like a button that says "advanced" or something. It may even require a UI redesign. 
Adding the online update button confused people enough, adding a managed mode WiFi wizard will just confuse them more.

I think it may also be about time for a complete rearrange of the settings menu to have an advanced settings tab.

justinmerrell, 2021-02-24:
I would highly recommend getting the installation process working on a clean version of Linux first. I think there is a lot of networking jank that has come from the OS img that keeps getting recycled. 

I agree with everything else

objorkman, 2026-10-13:
Network Settings (Settings → Advanced → Network) now has a mode selector with AP Mode, Client Mode, and Wifi Off. The choice is saved as `MODE` in `/home/kipr/wombat-os/configFiles/wifiConnectionMode.txt` and restored at boot (`NetworkManager::init()`), and AP mode is still the default.

closed 2026-10-13 by objorkman