kipr/botui #64: Adding C++ functionality or switching to C++ instead of C

Zacharyprime, 2021-03-03:
This is the botui compilation code:
```C++
void ProgramsWidget::compile()
{
	QModelIndexList currents = ui->programs->selectionModel()->selectedIndexes();
	if(currents.size() != 1) return;

  	const QString name = m_model->name(currents[0]);

  	qDebug() << "compile clicked for " << name;

	const QString projectPath = botui::pathToKISS + name;

	const QDir includeDir(projectPath + "/include/");
	const QDir srcDir(projectPath + "/src/");

	QString binFilePath = projectPath + "bin/botball_user_program";

        QString compileCommand = "gcc -o " + binFilePath + " -lwallaby -lpthread -lm -I " + includeDir.absolutePath() + " " + srcDir.absolutePath() + "/*.c";
	qDebug() << compileCommand;

	QByteArray ba = compileCommand.toLatin1();
	const char *compileCommandC = ba.data();

	int ret = std::system(compileCommandC);

	qDebug() << "ret = " << ret;
}
```

The compiler is set to C directly by calling gcc. 
If we wanted to switch to C++ we could just change this line:
```
QString compileCommand = "gcc -o " + binFilePath + " -lwallaby -lpthread -lm -I " + includeDir.absolutePath() + " " + srcDir.absolutePath() + "/*.c";
```

to:
```
QString compileCommand = "g++ -o " + binFilePath + " -lwallaby -lpthread -lm -I " + includeDir.absolutePath() + " " + srcDir.absolutePath() + "/*.cpp";
```

There is probably going to be other conflicts that are caused by this change, harrogate would also need to be changed to reflect this. I am not sure exactly what in Harrogate would need to be changed at this time, at some point I or whoever solves this issue will need to trace the path the code takes when you hit compile. Harrogate doesn't have any comments so it takes time to process what anything is. I am 70% sure that Harrogate feeds into pcompiler and pcompiler automatically determines the correct compiler, but I don't think g++ is setup as an option for pcompiler and it will default to gcc.

Additionally, I am unsure if KIPR as an organization will want to switch to C++, I don't know if there is some reason they decided to choose C. I view C as a primitive version of C++ and I can't think of a reason why you would choose to make a program in C over C++ other than experience level/simplicity. At very least I think we should add the option of C++.

Zacharyprime, 2021-03-05:
Update:
I just realized that we removed the compile button in botui at some point. So it's possible the botui line is irrelevant to adding C++.

chrehall68, 2022-05-17:
Check out my PR in the harrogate repository! It allows for c or c++ (since this is should really be a harrogate thing and not a botui thing).

objorkman, 2026-10-13:
botui no longer compiles programs: the KIPR IDE on the Wombat compiles them, and botui only runs `bin/botball_user_program`. The compile button and `ProgramsWidget::compile()` path are hidden and disabled, so the language choice belongs to the IDE.

closed 2026-10-13 by objorkman