kipr/botui #82: Test/Get botui working with Qt6

Zacharyprime, 2021-07-13:
Since Qt6 is released now, the goalpost may have moved.

Once we get Qt5 to a stable state, we will want to see if it works with Qt6.

I don't think this will require much effort since Qt5 and Qt6 are similar, but we will have to wait and see how we end up updating botui to Qt5.

Progress on Qt5: https://github.com/kipr/botui/issues/67

Zacharyprime, 2022-02-18:
Qt5 now works. However the jump to Qt6 can wait as it's not super urgent right now.
I also don't suspect the change will be as difficult as Qt5 was since they are similar.

When the community seems to be mostly using Qt6 I'll come back to this hopefully.

objorkman, 2026-10-13:
botui now builds and ships on Qt 6: `CMakeLists.txt` uses `find_package(Qt6 ...)`, the Wombat package is built against Qt 6, and CI builds every PR against Ubuntu 24.04's Qt 6.4.

closed 2026-10-13 by objorkman