kipr/botui #94: OpenCV4 Support for Botui

ayush055, 2021-07-24:
Trying to add OpenCV4 support to Botui

Current issues

- Camera widget on Botui crashes upon opening
- Camera configuration in settings does not work

Link to fork: [https://github.com/ayush055/botui/tree/opencv4-support](url)

Zacharyprime, 2021-09-09:
When updating to versions past jessie, the compiler throws warnings about the fact that we are using such an outdated opencv.

Zacharyprime, 2022-05-11:
Moving to openCV4 is now a required task to get a WombatOS that we can build from scratch.

Erin and I have already started working on porting it.
As far as I've seen, all the functions we use still exist in CV4 so it shouldn't bee too bad.

Zacharyprime, 2022-05-16:
I'm going to place the issues ayush mentioned above in a new issue because I do not think they are related to the version of opencv. I tested this on OpenCV4 and it still exists.

closed 2022-05-16 by Zacharyprime