kipr/botui #111: Bug in `NetworkSettingsWidget::updateInformation`

bmcdorman, 2023-01-06:
The following bug exists in `NetworkSettingsWidget`. A fix was implemented that complicates botui's `RootController`. This erroneous fix needs to be removed and the underlying bug corrected in the `erinQt6Branch`.

https://github.com/kipr/botui/blob/55fdc58b0ee93085d0a2f62d7b96c5ad1ff2ea0f/src/NetworkSettingsWidget.cpp#L119

chrehall68, 2023-05-23:
The fix you mentioned was removed in eab3ad8ff471bef6be600596cb84b33c3ab53b62.

chrehall68, 2023-06-02:
However, the underlying logic of checking the connectionModeSelect button remained in erinQt6Upgrade, causing #130 . This is now fixed in #131

chrehall68, 2023-06-06:
Both the line and the underlying logic have been fixed, this issue can be closed.

objorkman, 2026-10-13:
#131 was closed without merging, but the code this issue describes has since been replaced on master: eab3ad8 removed the `RootController` workaround from 55fdc58 and rewrote `NetworkSettingsWidget::updateInformation()`, which now sets `connectionModeSelect` from NetworkManager's state instead of reading it first. Reopen if the bug shows up again.

closed 2026-10-13 by objorkman